JChristensen/JC_Button #3: Double-click

atuline, 2017-02-11:
I've used this library for a few of my FastLED routines over the past couple of years and would love to see a double-click feature added, so that it would support:

- click
- double-click
- long click

Anyways, thanks for the library.

JChristensen, 2017-02-14:
Thanks for the suggestion. I've thought about it but haven't really had a need. I'm not doing much development of this sort currently but will keep it in mind for the future.

atuline, 2017-02-14:
No problem. I come and go myself on this sort of thing myself.

closed 2017-02-14 by atuline

marmilicious, 2020-07-04:
@JChristensen  Just thought I'd give a quick ping to see if you had giving any more thought to adding a double click option?  It would be a nice addition that would really round out your button library.  

Thank you.
-marc

JChristensen, 2020-07-04:
No, sorry. I've been too busy with other things these days.

marmilicious, 2020-07-04:
Totally understand.  
Thank you for your recent updates last year.