JChristensen/JC_Button #7: btn.pressedFor(ms) makes the code run repeatedly

giogziro95, 2017-12-12:
I have something similar to this:

```
void loop() {
  if (btn.pressedFor(500)) {
    // Do stuff
    Serial.println("btn pressed for 500 ms");
  }
}
```

When I hold down the button for 500 ms, the serial monitor shows that it runs multiple times, and I see why in the library code. However, what I'm looking for is one of the following:

1. When I hold it down, it should run once and do not run again until I release it and hold it down again.
2. When I hold it down, it should run periodically with the 500 ms intervals until I release it. (Prefered)

Is there an easy way to implement one of these in my code without messing with the library itself?

JChristensen, 2017-12-12:
I find it helpful to think about the state the button is in and also what state the user is in. The trick is in understanding how the user will react and having the code respond appropriately. For (1), once I hold the button for 500ms, then I will see the message. Once I see the message, at some later time I will release the button. So, the code has to wait for me to release the button. This requires another test and continued polling of the button. Here is a simple example, but for more complicated situations when there are other things the code has to do, a finite state machine will probably be more appropriate.

```c++
void loop()
{
  btn.read();                    //read the button
  if (btn.pressedFor(500)) {
    // Do stuff
    Serial.println("btn pressed for 500 ms");
    while (btn.isPressed()) btn.read(); //wait for user to release the button
  }
}
```
For (2) see the [UpDown example](https://github.com/JChristensen/Button/blob/dev/Examples/UpDown/UpDown.pde).

closed 2017-12-12 by JChristensen

giogziro95, 2017-12-12:
Thanks for the response. Unfortunately, the while loop would result in blocking scenario which doesn't work for me.

JChristensen, 2017-12-12:
Then that's where I'd use a state machine.

giogziro95, 2017-12-14:
I also have multiple buttons so the code was getting too convoluted, so decided to solve it by modifying the library and adding new methods. The fork can be found [here](https://github.com/giogziro95/Button). If you think the changes fit into the project purposes, I'll create a PR against the _master_ branch (since _dev_ seems to be inactive).

giogziro95, 2017-12-19:
@JChristensen Hey, shall I do a PR? Any suggestions?

telmomarques, 2025-01-11:
The "is"/"was" pattern exists for simple press or release, why not for long press?

It's stated on the readme that the "was" functions are most useful, and I'd argue that's the primary reason for someone to use this library, so I was a bit surprised not finding the same pattern for long presses.

telmomarques, 2025-01-11:
Hi @JChristensen, thanks for the reply!

> How shall we distinguish the "was" condition from the "is" condition?

From the readme:
_(...) they actually detect a change in the state of the button, which is usually what we want in order to cause some action._

Maybe I'm missing something here, but it'd be the same as for `isPressed()` and `wasPressed()`:

- `isPressedFor(int ms)` would be true for as long as the button <ins>is being</ins> pressed (after elapsed ms)
- `wasPressedFor(int ms)` would be true <ins>once</ins>, when the change is detected (after elapsed ms)

It's exactly the same behavior as for a press, but for a long press.

> from an application perspective, what are we trying to accomplish?
A non-blocking way of detecting the change that represents a long-press, and running an action once when that change happens (instead of on every cycle).

telmomarques, 2025-01-11:
I just noticed the intended behavior can be achieved like this:

```
Button longPressButton = Button(GPIO_NUM_27, 2000); // 2000ms debounce to "simulate" a long press
if(longPressButton .wasPressed())
{
  // Will run once
}
```

But it would be nice to achieve this behavior without having to create a separate button instance.

JChristensen, 2025-01-11:
Applying a couple techniques can achieve this:
(1) After a long press, wait for the release event, and
(B) Use a state machine to avoid blocking.

For example:
```c++
#include <JC_Button.h>

Button btn(7);  // button on pin 7

void setup()
{
    Serial.begin(115200);
    btn.begin();
}

void loop()
{
    enum states_t {WAIT_PRESS, WAIT_RELEASE};
    static states_t state {WAIT_PRESS};

    btn.read();
    switch (state) {
        case WAIT_PRESS:
        if (btn.pressedFor(2000)) {
            state = WAIT_RELEASE;
            Serial.println("pressed");
        }
        break;
    
        case WAIT_RELEASE:
        if (btn.wasReleased()) {
            state = WAIT_PRESS;
        }
        break;
    }
}
```

telmomarques, 2025-01-11:
Thank you for the alternate solutions, but the question is why doesn't the lib provide this behavior.
Seems like the value for running a single action on single press is acknowledged, but for not for long press.

JChristensen, 2025-01-11:
I am trying to understand.

This is exactly the behavior of pressedFor():

* `isPressedFor(int ms)` would be true for as long as the button is being pressed (after elapsed ms)

The "once" is confusing me here. What is the definition of once? Only a single time, until a hardware reset? 

* `wasPressedFor(int ms)` would be true once, when the change is detected (after elapsed ms)

telmomarques, 2025-01-11:
> The "once" is confusing me here. What is the definition of once? Only a single time, until a hardware reset?

Sorry, I'll try to give more detail.
By "once" I mean every state change, the same definition as for `wasPressed()`.
In other words, the same behavior as state machine example provided above.

JChristensen, 2025-01-11:
Remember the overriding assumption with this library is that `read()` is called frequently. This is necessary to be responsive to user button inputs.

Say we have a hypothetical function, `wasPressedFor()`.
We call `wasPressedFor()` and it returns true, so we take some action. The next time we call `read()`, the user still has the button pressed (it will take many milliseconds for them to release it.) If we now call `wasPressedFor()` again, it must return false, because it has already returned true "once." If we continue to call `wasPressedFor()`, under what circumstance can it again return true?

telmomarques, 2025-01-11:
> under what circumstance can it again return true?
Under the circumstance that it was released and long pressed again, except for the first long press since a hardware reset.
Releasing the button would clear the flag, enabling another "once" action to be performed on long press.

nerdius, 2025-01-12:
This was 8 years ago, but here's my implementation:

```
/*----------------------------------------------------------------------*
 * wasPressedFor(ms) and wasReleasedFor(ms) are the same as             *
 * pressedFor(ms) and releasedFor(ms), except that when the button is   *
 * pressed (or released) for the specified time in milliseconds,        *
 * instead of repeatedly returning true (1), it returns the value in    *
 * the time intervals of the specified time in milliseconds.            *
 * These functions do not cause the button to be read.                  *
 *----------------------------------------------------------------------*/
uint8_t Button::wasPressedFor(uint32_t ms)
{
    if (_state == 1 && _time - _intervalStart >= ms) {
        _intervalStart += ms;
        return 1;
    }
    else {
        return 0;
    }
}
uint8_t Button::wasReleasedFor(uint32_t ms)
{
    if (_state == 0 && _time - _intervalStart >= ms) {
        _intervalStart += ms;
        return 1;
    }
    else {
        return 0;
    }
}
```

[Commit](https://github.com/nerdius/Button/commit/2c97c541d7a1d9cd8b35d0fb176625e6080d3060)

@telmomarques, I believe you want this behaviour. My fork is unmaintained, but you can always patch the latest version from @JChristensen.

nerdius, 2025-01-12:
This was my use case:

Short press would increment/decrement the value of the countdown timer (and the corresponding display) by one minute, a long press for one second would increment/decrement it by 10 minutes. If you kept holding the button for longer, it'd keep incrementing it by 10 minutes in every 0.1 seconds. The delays could be adjusted in the code, obviously. The goal was to quickly set and modify the timer.

JChristensen, 2025-01-12:
> This was my use case:
> 
> Short press would increment/decrement the value of the countdown timer (and the corresponding display) by one minute, a long press for one second would increment/decrement it by 10 minutes. If you kept holding the button for longer, it'd keep incrementing/decrementing it by 10 minutes in every 0.5 seconds. The delays could be adjusted in the code, obviously. The goal was to quickly set and modify the timer.

See [this example](https://github.com/JChristensen/JC_Button/blob/master/examples/UpDown/UpDown.ino).

JChristensen, 2025-01-12:
> > under what circumstance can it again return true?
> 
> Under the circumstance that it was released and long pressed again (except for the first long press since a hardware reset, no need to wait for a release in that case). Releasing the button would reset the state, enabling another "once" action to be performed on long press.

The rub here is that the action needs to happen immediately once the long-press interval is met. This usually generates some feedback to the user that causes them to release the button. Building this into the library might then require a callback, etc.

This feels like we are trying to make the library do the application's work. The library does not know about events at the application level.

I am still struggling to understand what problem we are trying to solve. As ever, I am willing to be convinced, but it will take a good complete problem statement and probably a complete (if simple) code example.

nerdius, 2025-01-12:
> See [this example](https://github.com/JChristensen/JC_Button/blob/master/examples/UpDown/UpDown.ino).

I don't remember the exact details, but I also had display multiplexing, which made it impossible to use a simple loop (because of blocking). I ended up modifying the library in order to make the code simpler and more readable.