JChristensen/JC_Button #8: SDK update issue

chriscamacho, 2018-02-08:
after cloning to Arduino/libraries i've had to rename to Button-git as if left it get over written next time there is a libraries update - either a different lib call Button or a very early version of Button?

either way it breaks my code!

JChristensen, 2018-02-08:
Hi, I'm not sure I completely understand. I am aware that there is a different library that has the same name registered with the Arduino Library Manager. Installing that library should be optional. If you have it installed, then I'd think uninstalling it should avoid any issues.
If you can provide specific steps to recreate the issue, then I may be able to provide further guidance.

chriscamacho, 2018-02-08:
If I have this library installed, next time the sdk ask to do updates it over writes this library with the other one with the same name, I have never installed the other library with the same name, a simple work around is to rename the folder, the library name can stay the same (as long as you don't install the other library)

JChristensen, 2018-02-08:
Thanks, that helps. My Button library doesn't have a library.properties file. Adding one would be easy enough, I'll try it. But if the library manager works strictly off the name then that may not help. I'll get back to you.

JChristensen, 2018-02-08:
I added the library.properties file. Try reinstalling my library and see if the issue goes away. I have a test machine with a vanilla version of Arduino 1.8.5. I installed my library there (from the .zip file downloaded from GitHub) and it's not showing up in the list of Updatable libraries so I think that has fixed the issue.

JChristensen, 2018-02-19:
Nothing heard; assuming fixed by 3b10b31c5fc59190566b8d0e5fd69eb624b91bec.

closed 2018-02-19 by JChristensen