JChristensen/JC_Button #12: Library manager

scls19fr, 2018-05-05:
Hello,

I noticed that this library have a [library.properties](https://github.com/JChristensen/Button/blob/master/) file. Although it doesn't seem to be published to Arduino library manager.
https://github.com/arduino/Arduino/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+label%3A%22Component%3A+Board%2FLib+Manager%22+

Kind regards

JChristensen, 2018-05-05:
There is a different library by the same name already in the library manager. I assume there cannot be two different libraries by the same name. Do you have a recommendation?

JChristensen, 2018-05-05:
Well that's interesting. Last I checked I thought there was a "Button" library already in library manager. I just downloaded the index and I don't see it now. Maybe I'll submit this one then. Thanks.

JChristensen, 2018-05-05:
No, scratch that. It's still there.

closed 2018-05-05 by JChristensen

scls19fr, 2018-05-06:
Is there a website where all Arduino libraries available to library manager can be found?

JChristensen, 2018-05-06:
The index file is http://downloads.arduino.cc/libraries/library_index.json

JChristensen, 2018-05-11:
@scls19fr: FYI, I've made a fairly thorough update. The library and repo have been renamed to JC_Button. I've requested that it be added to the Arduino library manager but this usually takes a few days. [The latest release](https://github.com/JChristensen/JC_Button/releases/tag/2.0.0) makes changes to the constructor that are not backwards-compatible and also adds a `begin()` function.

scls19fr, 2018-05-11:
Thanks @JChristensen 
The decision of changing license is your choice and I respect it.
However I prefer LGPL BSD or MIT license for libraries.

JChristensen, 2018-05-11:
I'm certainly not an expert on licenses. I'd be interested to know the thinking behind your preferences.

scls19fr, 2018-05-11:
When licensing a library under GPL you "force" user of your library to adopt the same license. As a developper I don't consider this as freedom and always prefer libraries with licenses which give me the choice of what license I will choose for my own projects.

I won't send you links about differences between open source licences, they are often biased depending of what author of article consider as more important.

JChristensen, 2018-05-11:
Thanks. I have done a little reading and that was my understanding of GPL. I'm impressed with the open source movement and I like to think I'm giving some back even if in a small way. I can't admit to having thought deeply about licensing, but I'm only doing hobby grade stuff, so I figure there's not a lot at stake.
Thanks again, great chatting with you!

scls19fr, 2018-05-11:
I've just published (under MIT license) a Python Pandas script to search for Arduino libraries https://github.com/scls19fr/arduino_libraries_search the number of button / debounce library is quite high... although currently Arduino doesn't provide information about license ...
I will probably improve this script to crawle website to search for license... that's an other way to "give some back even if in a small way" 😄
Thanks also