JChristensen/JC_Button #23: Not working with Nano Every

barnosch, 2019-06-25:
I get this error while uploading on a brandnew Nano Every, and the Buttons are not working.
Maybe you can have look into this?
WARNUNG: Bibliothek JC_Button behauptet auf (avr) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (megaavr) Architektur(en) ausgeführt wird.

davidmpye, 2019-06-25:
Wouldn't worry about it.

It is just the library properties file only specifies avr not megaavr etc...

JChristensen, 2019-06-25:
@barnosch, I'm not clear on exactly which board that is. Would you please provide a link?

JChristensen, 2019-06-25:
@barnosch, I wouldn't worry just about the warning, but if buttons are not working, then there may be an issue. I was not aware of the Nano Every, I see that it is a new board. I do not have a Nano Every so I cannot attempt to reproduce the issue. Do any of the example sketches from the library work?

barnosch, 2019-06-26:
@JChristensen thanks for the quick reply.
I could get the "UpDown" example to work. So this works. 
"SimpleonOnOff" throwed a error `cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'` but with a quick and dirty `digitalWrite(LED_PIN, 1);` i could at least get the LED to light up (once). 
Why my common sketch does not work anymore with the JC_Buttons, with the new [Nano Every](https://store.arduino.cc/nano-every) i am not sure yet.

JChristensen, 2019-06-26:
Hello @barnosch,

Thanks, I can reproduce that compile error. Evidently the second argument in `digitalWrite()` is required to be a `PinStatus` type. That seems overly restrictive to me and is inconsistent with other board types but I don't know why that decision might have been made.

I can work around the compile error with static_cast but I cannot test it, not having a board.

Would you please try the following with the SimpleOnOff example and let me know what happens?

In the SimpleOnOff example, change [line 32](https://github.com/JChristensen/JC_Button/blob/master/examples/SimpleOnOff/SimpleOnOff.ino#L32) from:
```c++
digitalWrite(LED_PIN, ledState);
```
to:
```c++
digitalWrite(LED_PIN, static_cast<PinStatus>(ledState));
```

barnosch, 2019-06-26:
Thanks for the quick fix. That works perfectly.

Maybe you can search for the "Arduino megaAVR Boards" library in the boards administration.  I was advised after pluggging in the new Nano Every to install that to get support from the IDE

JChristensen, 2019-06-26:
Yes, I installed "Arduino megaAVR Boards" with the boards manager which is how I was able to do a compile for the Nano Every. I was just not able to actually test that the code worked. Thanks!

closed 2019-06-26 by JChristensen

seisfeld, 2019-06-28:
Maybe this issue needs to reopened. While the SimpleOnOff example works fine (given one applies above static_cast) on the Arduino Nano Every when using the D pins, it does not when using the A Pins. Like A0. If you do so, the button presses are not registered. AFAIK the A Pins are supposed to be usable as GPIOs. Any clue why this does not work? It does work fine on a regular Nano.

JChristensen, 2019-06-28:
@seisfeld, just so I'm clear, you are running the SimpleOnOff example with the following changes only?
1. BUTTON_PIN(A0);
2. digitalWrite(LED_PIN, static_cast<PinStatus>(ledState));

On the Tools menu, which Register Emulation setting is selected? Does changing that make a difference?

seisfeld, 2019-06-28:
@JChristensen Yes, correct. If works fine with `BUTTON_PIN(12);` (for example) but it does not work with `BUTTON_PIN(A0);` (A0-7 for that matter).

I tested both Register Emulation settings, and neither does work with any of the A-Pins.

JChristensen, 2019-06-28:
@seisfeld, thank you (I don't have a board to test with.)
I wonder if the internal pullup resistor is not getting enabled properly.
Would you try adding an external pullup resistor from the button pin (e.g. A0) to Vcc?
I'd use a 10K but value is not critical, anything between 1K and 100K should work.

seisfeld, 2019-06-28:
No problem. I quickly added a 10K between A0 and the 5V Pin. No change.

JChristensen, 2019-06-28:
Interesting. Time for some basic testing. Try the following sketch and observe the output as the button is pressed and released. Should show "0" while the button is pressed, and "1" when it is not.
```c++
uint8_t testPin(A0);

void setup()
{
    Serial.begin(115200);
    Serial.println("Start");
    pinMode(testPin, INPUT_PULLUP);
}

void loop()
{
    Serial.print(millis());
    Serial.print(' ');
    Serial.println(digitalRead(testPin));
    delay(1000);
}
```

seisfeld, 2019-06-28:
When using A0 it always displays 1. When using say Pin 10 it toggles as expected.

JChristensen, 2019-06-28:
Wow, that is very interesting! Do you use the Arduino forum? I'd say post the test sketch there and see is someone else can confirm your results. If so, then raise an issue with Arduino.

JChristensen, 2019-06-28:
I'd also be curious to know whether the basic Blink example sketch works as expected on pin A0. If not, that might be further evidence of a pin mapping issue.

seisfeld, 2019-06-28:
No, it does not. LED stays off. Again on Pin D10... works as expected.

barnosch, 2019-06-28:
Can confirm that. For me it is the same.

JChristensen, 2019-06-28:
Thanks to both of you. I'm surprised that the Arduino forum is not full of posts about this. Maybe it's early yet, not sure how many of these boards are in use.
I'd say opening an issue at https://github.com/arduino/Arduino would be appropriate. Would one of you want to do that? If not, I could do it and include your names as having verified the issues.

seisfeld, 2019-06-28:
I can do that, was about to ask you which of the gazillion repos of the arduino org is appropriate for that. Thanks for your help so far Jack!

JChristensen, 2019-06-28:
Guess I'm not 100% sure about which repo, it could also be https://github.com/arduino/ArduinoCore-API
Pick one, and you'll probably hear if it's wrong!
Glad to help, it's always fun to find fundamental errors like that. Sounds like QC may have been a bit lacking on that new board.

JChristensen, 2019-06-28:
Also be sure you are both using the latest versions, Arduino IDE 1.8.9 and the Arduino megaAVR Boards 1.8.1 as that will probably be the first question. Best to include that information in the issue.

seisfeld, 2019-06-28:
Yep. I have both of these installed and I have it listed right at the top of the issue.

seisfeld, 2019-06-28:
Here is the issue for reference: https://github.com/arduino/Arduino/issues/9024

JChristensen, 2019-06-28:
Looks great, should be interesting to see what the answer is.
https://github.com/arduino/Arduino/issues/9024

seisfeld, 2019-06-28:
Indeed... :)

seisfeld, 2019-06-28:
There was an update to the issue already. It's a confirmed bug, and it was discovered just after the megaAVR 1.8.1 release. It's already fixed in master and the fix will be included in the 1.8.2 sometime next week. For details, see here: https://github.com/arduino/Arduino/issues/9024#issuecomment-506777330

I have been given a zip with a beta and if I install that, JC_Button works as expected - even in A0. ;)

JChristensen, 2019-06-28:
Excellent, thanks for the update. Seems like quite a blooper. My expectation would be that proper pin mapping is tested prior to release.

seisfeld, 2019-06-28:
I'm happy it's fixed though. But you are right, it's a little embarassing. 😇 On the other hand, shit just happens.

seisfeld, 2019-07-01:
megaAVR-1.8.2 has just been released. So the A-Pins work again. Are you planning to officially make JC_Button available for that core?

JChristensen, 2019-07-01:
My policy is to indicate in the library.properties file only those architectures that I personally use and have tested. Right now that is limited to the AVR architecture and will probably not change any time soon. If the warning message bothers people, it's trivial to change the library.properties file to contain `architectures=*`

seisfeld, 2019-07-01:
I understand. It's a little unfortunate that its actually listed as "INCOMPATIBLE" (yes in caps) in the Arduino IDE, but well that's how it is then. I'm glad it works so far anyways. :)

Thanks again for you kind assistance.

davidmpye, 2019-07-01:
As I've said before though, there is one downside to this.

Not because of the arduino IDE, but because of other IDEs like Platformio.
Platform.io has a library dependency finder which will automatically
download and install the libraries for you.  But on ESP8266 (which I use)
it won't resolve and download JCButton because it only says AVR.     This
is really a library which would benefit from generic, as it doesn't use any
platform specific code, so if it doesn't work on a platform, the problem is
the platform's (as this thread has nicely demonstrated...!)

Ultimately, it's JC's decision, as it's his library, but it is a bit
unfortunate for these reasons.

David

JChristensen, 2019-07-01:
Very odd; as of right now, the Nano Every has disappeared from my Tools > Board menu and the megaAVR library doesn't even show in the Boards Manager. Can either of you confirm that?

JChristensen, 2019-07-01:
Regarding the other points. The message I get has WARNING in caps and says "_may_ be incompatible". I wonder if perhaps it looks different on a different OS or IDE, etc. Verbatim from the console section of the Arduino IDE:
```
WARNING: library JC_Button claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (megaavr) architecture(s).
```
As for other IDEs, etc. I just do not have the bandwidth or in many cases the interest. It seems that new boards are being released faster than ever; I just cannot keep up. I was not even aware of the "Nano Every" (what a name!) until this issue was raised. This is a (very) part-time hobby for me and my goal is not to be all things to all people (or to all boards!) So by only indicating support for architecture(s) that I work with regularly, I feel I am saying that support for other architectures may not be a priority. This issue _was_ a good illustration and I don't even mind working on this type of issue and I have made accommodations in some of my libraries for other architectures. But it all comes back to the bandwidth thing.

Thanks for understanding, I've enjoyed working with you guys. It's a great community with a lot of good folks.

seisfeld, 2019-07-01:
> Very odd; as of right now, the Nano Every has disappeared from my Tools > Board menu and the megaAVR core doesn't even show in the Boards Manager. Can either of you confirm that?

No, I just uninstalled and reinstalled the megaAVR 1.8.2 without issues.

As for the _incompatible_ thing, I was referring to the blow (i.e. when you want to open one of the supplied examples):

[image]

But I get your point, there is just not enough bandwidth to support all and every board which is perfectly fine. And if one wants to use JC_Button on the Every (maybe because it's required by a project) it does work, despite the warning. So for the time being I think everyone can live with that. :)

cheers
Stephan

JChristensen, 2019-07-01:
Ah right, thanks. Yes I've seen that. Seems rather heavy-handed; the examples _may_ be incompatible. Seems like the compile time warning should be enough. Not sure why I don't see the Nano Every any more but I'll assume it's a personal problem with my installation.