JChristensen/JC_Button #24: Would this be quick enough for the debounce on a reed switch flow meter?

mrarmyant, 2019-07-17:
What’s the max frequency this can handle?

JChristensen, 2019-07-17:
A very simple sketch that doesn't do much else can poll a switch in excess of 70kHz when running on a 16MHz AVR microcontroller like an Arduino Uno.

But at some point I assume there is other work to do, something needs to be done with the data, etc. So a better question may be how long does that take.

Depending on the expected frequency generated by the reed switch flow meter, and how long other processing takes, etc., I might opt for hardware debouncing and use the reed switch to drive an interrupt, especially if I were concerned about missing a switch closure.

Hope this helps.

JChristensen, 2019-07-17:
Maybe a better way to think about it is that by definition the maximum switch frequency is the reciprocal of the debounce time. So for example, if I have a 20mS debounce time, then the switch can't operate faster than 50Hz.

Then the question becomes, can whatever other processing that is needed occur in less than the debounce time.

mrarmyant, 2019-07-17:
I guess I was hoping this was on an interrupt pin but at the same time it might never get to send data out if that was the case.

closed 2019-07-17 by mrarmyant

JChristensen, 2019-07-17:
The library will work on any pin, it just polls the switch. I pretty much only intended it for HMI or similar situations. Specialized sensors probably deserve their own coding in many cases.