JChristensen/JC_Button #25: no syntax highlighting when library is included

thoschneider, 2019-10-02:
Hello!
Unfortunately there is no syntax highlighting when the library is included. I think adding the line "JC_Button Keyword1"
to keywords.txt does the trick. 

(As I read "Keyword1" should be used for classes. You named your class "Button", but the name of your libary is "JC_Button". Without adding "JC_Button Keyword1" to keywords.txt there is no syntax highlighting when library is included.)

JChristensen, 2019-10-02:
Not sure I'm understanding. What specifically is not highlighted? Here is what I see for one of the example sketches:

[image]

thoschneider, 2019-10-03:
Dear Mr. Christensen,

maybe it is a coincidence, but in all the sketches I used so far, the included 
header file was highlighted in orange (because the name of the header file 
matched a class name in that file). See screenshot attached.

Me and my students found this highlighting very convenient indicating that the 
library is (properly) installed. Adding "JC_Button KEYWORD1" to keywords.txt 
would suit this purpose.

It is not a big deal, just a naming convention perhaps. I hope you get my 
point.

Best regards

Thomas

Am Donnerstag, 3. Oktober 2019, 00:12:01 CEST schrieb Jack Christensen:
> Not sure I'm understanding. What specifically is not highlighted? Here is
> what I see for one of the example sketches:
> 
> ![SimpleOnOff](https://user-images.githubusercontent.com/762571/66085715-010
> fef80-e540-11e9-856d-c442240c0aa8.png)

JChristensen, 2019-10-03:
Hello Thomas,

Your screenshot doesn't seem to be showing for me but it sounds like you are just talking about the #include statement. Adding the line to the keywords file as you suggest should fix that. Look for it in a day or two.

Cheers ... Jack

thoschneider, 2019-10-03:
Yes, only the #include statement.

Thomas

Am Donnerstag, 3. Oktober 2019, 03:35:45 CEST schrieb Jack Christensen:
> Hello Thomas,
> 
> Your screenshot doesn't seem to be showing for me but it sounds like you are
> just talking about the #include statement. Adding the line to the keywords
> file as you suggest should fix that. Look for it in a day or two.
> 
> Cheers ... Jack

JChristensen, 2019-10-03:
I'm just curious, you mentioned your students. What level are you teaching at?

closed 2019-10-03 by JChristensen

thoschneider, 2019-10-09:
> thread [2].
> Links:

I offer some arduino courses in an "MINT"-association 
(math,informatics,natural sciences) for kids and teens from 12-16 years. 
I am a highschool teacher for physics and biology and since three years 
I sporadically offer some arduino projects.

Cheers,
Thomas

JChristensen, 2019-10-11:
Very cool. I've tried that once or twice on a very small scale with friends' kids, but I am not a teacher so I'm not too good at it!

commit f379405f (2019-10-03): Add JC_Button to keywords file so that syntax highlighting works on #include statement. Closes #25.