JChristensen/JC_Button #26: Detect Simultaneous Events

JacksonAudio, 2020-07-27:


JChristensen, 2020-07-28:
The trouble lies in what we consider "simultaneous". If I run the following sketch on an Arduino Uno, the loop executes nearly 40,000 times per second. Since we are sampling the buttons that fast, "simultaneous" means that they both have to be released within about 25 microseconds of each other. Even if this were humanly possible, normal mechanical variation in the buttons makes it quite unlikely. Even the bounce time of a typical button switch might be two or three orders of magnitude larger than the sample time.
```c++
#include <JC_Button.h>
Button btn1(7), btn2(8);

void setup()
{
    btn1.begin();
    btn2.begin();
    Serial.begin(115200);
}

void loop()
{
    btn1.read();
    btn2.read();

    if (btn1.wasReleased() && btn2.wasReleased()) {
        Serial.println("both released");
    }
    else if (btn1.wasReleased()) {
        Serial.println("btn1 released");
    }
    else if (btn2.wasReleased()) {
        Serial.println("btn2 released");
    }
}
```
One solution would be to sample the buttons at a slower rate. The sketch below samples 20 times per second (i.e. every 50 milliseconds) and is much better at detecting a "simultaneous" release. Hope this helps.
```c++
#include <JC_Button.h>
Button btn1(7), btn2(8);

void setup()
{
    btn1.begin();
    btn2.begin();
    Serial.begin(115200);
}

void loop()
{
    static uint32_t msLast;
    uint32_t ms = millis();
    if (ms - msLast >= 50 ) {
        msLast = ms;
        btn1.read();
        btn2.read();
    
        if (btn1.wasReleased() && btn2.wasReleased()) {
            Serial.println("both released");
        }
        else if (btn1.wasReleased()) {
            Serial.println("btn1 released");
        }
        else if (btn2.wasReleased()) {
            Serial.println("btn2 released");
        }
    }
}

JacksonAudio, 2020-07-28:
Thank you so much for the fast response! I'll check this out now!

JacksonAudio, 2020-07-28:
`if (millis() > timer_slow_stuff + 1000/20) { // put the slow updating stuff here
  swDist.read();  // Read the button
  swDrive.read(); // Read the button

  int A = swDist.wasReleased();
  int B = swDrive.wasReleased();

  int allBtnStates = A + 2*B;
  if (allBtnStates < 4) { // Any button pressed
      switch (allBtnStates) {
        case 3: Serial.println("Both A and B pressed"); break; // Both A and B pressed
        case 2: Serial.println("Only A pressed");       break; // Only A pressed
        case 1: Serial.println("Only B pressed");       break; // Only B pressed
        // case 0: Serial.println("NONE pressed"); break; // NONE pressed
      }
  }

  timer_slow_stuff = millis();
}`

JacksonAudio, 2020-07-28:
Sorry I can't get my code to look as good as yours. 

This code does do the job pretty well. I "borrowed" it from here: "https://stackoverflow.com/questions/53900441/multiple-button-press-management-in-arduino"

JChristensen, 2020-07-28:
That code may appear to work OK but it will stop working in about 49 days when millis() overflows. Using subtraction rather than addition will avoid this, e.g.:
```c++
if (ms - msLast >= 50 ) {
```

closed 2020-07-28 by JChristensen

JacksonAudio, 2020-07-29:
Very smart. :) 

> That code may appear to work OK but it will stop working in about 49 days when millis() overflows. Using subtraction rather than addition will avoid this, e.g.:
> 
> ```c++
> if (ms - msLast >= 50 ) {
> ```

JacksonAudio, 2020-07-29:
I've almost got this thing perfected. The only issue i'm having is that when you do a long press (swDrive.pressedFor(1000) when the switch is released it calls the wasReleased routine. Is there a way to selectively bypass that? Here's the whole bit of code: 

`if (millis() > timer_slow_stuff + 1000/20) { // put the slow updating stuff here

    swDist.read();  // Read the button
    swDrive.read(); // Read the button

    if (swDrive.pressedFor(1000)) {
      switchCase = 5;
    } else if (swDist.pressedFor(1000)) {
      switchCase = 4;
    } else if (swDist.wasReleased() && swDrive.wasReleased()) {
      switchCase = 3;
    } else if (swDrive.wasReleased()) {
      switchCase = 2;
    } else if (swDist.wasReleased()) {
      switchCase = 1;
    }

    switch (switchCase) {
      case 5:     processDriveHold();
                  switchCase = 0;
                  break;
      case 4:     processDistHold();
                  switchCase = 0;
                  break;
      case 3:     readBothFootSwitches();
                  break; // Both A and B pressed
      case 2:     readDriveFootSwitch();
                  break; // Only A pressed
      case 1:     readDistFootSwitch();
                  break; // Only B pressed
    }

    timer_slow_stuff = millis();

  }`

JChristensen, 2020-07-29:
I really prefer to limit issues here to actual problems with the library as opposed to fixing user code. That said, it's just a logic problem in your code. If you want to react to a long press but not a release, then don't test for a release. A finite state machine might be a good choice but the implementation with the big if statement outside of the switch seems unusual and hard to follow. I like to start by drawing a state transition diagram. Once I have that correct, the code just about writes itself.