JChristensen/JC_Button #28: Explicit set to togglebutton

Winfried-Bantel, 2020-12-09:
For my sonoff s-20 i extend the toogle-button, because via mqtt excplicit "swotch-on" or "switch-off" request is received, and in this case the togglebutton is in the wrong state.

		bool toggleState(bool s) {return m_toggleState = s;}

The actual state is corrected and returned.
Now:
void switch_via_mqtt(int value) {
    toggleState(value);
}

JChristensen, 2020-12-09:
My apologies, I do not understand. Is there an issue with the library? Please try to explain the issue outside the context of your project. Ideally, I would like a problem statement and a short, complete code example that doesn't require a bunch of other libraries (e.g. MQTT or whatever.)
Thanks.

JChristensen, 2020-12-09:
Read your issue a couple more times. Are you saying that the button controls the state of some other object, but that object's state is also controlled by another process that the Button library is unaware of (e.g. MQTT) and therefore there is a need to be able to force the state of the ToggleButton object, even when the button hasn't actually been pressed?

JChristensen, 2020-12-19:
Nothing heard. Closing issue.

closed 2020-12-19 by JChristensen