JChristensen/JC_Button #36: Compilation warning when working with Arduino Rev4

gstavrinos, 2023-09-30:
The specific warning is this one:
`WARNING: library JC_Button claims to run on avr architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).`

Are there any known problems? Can the warning be **safely** ignored?

If there are concerns, I would be happy to help out, if I can.

Thanks for any reply and your work so far with the library!

davidmpye, 2023-09-30:
The issue is that the author hasnt marked the library as compatible with
that architecture (and has chosen not to do so).

The library contains no platform-specific code.

David

JChristensen, 2023-09-30:
> Are there any known problems? Can the warning be **safely** ignored?

Unknown. As noted, the library does not contain platform specific code, so my gut feel is that it should be OK but there are no guarantees. My work is pretty much limited to the AVR architecture; I cannot keep up with all the new boards and architectures. I am a bit reticent to offer my code as approved for hardware that I have not personally tested and cannot debug any issues on.

If the warning bothers you, in the library.properties file, just change
```architectures=avr```
to
```architectures=*```

Thanks for the kind words and good luck with your project.

closed 2023-09-30 by JChristensen